jadonk/beaglebone-blue #28: Are alternate colors available for the JST connectors?

jadonk, 2016-10-12:
This would be to assist in new users locating specific connectors.

erikwelsh, 2016-11-01:
Here is the color scheme for the connectors:

2 pin
M1-4: (4x) white

3 pin
DSM2: (1x) white

4 pin
E1-4, I2C: (5x) white
UT0, UT1, UT5: (3x) black
PWR: (1x) new color #1 (red preferred)
CAN: (1x) new color #2 (green preferred)

6 pin
S1.1, S1.2: (2x) white
GP0-1: (2x) black
ADC: (1x) new color #1 (red preferred)
GPS: (1x) new color #2 (green preferred)

Unfortunately, the colored connectors are a 50% premium over the other connectors so one goal of the above scheme was to minimize the number of colored connectors.

jadonk, 2016-11-15:
If the only secondary color we have is black, let's use white on everything except using black on PWR and ADC. Those are the ones most likely to damage something if connected mistakenly.

cstiegman, 2016-11-15:
Got it - thanks!

Regards,
Craig Stiegman
element14
(773) 90<tel:(773)%20907-5941>7-5941<tel:(773)%20907-5941>

********************************Disclaimer***********************************
The preceding e-mail message (including any attachments) contains information 
that may be confidential, be protected by the attorney-client
or other applicable privileges, or constitute non-public information. It
is intended to be conveyed only to the designated recipient(s). If you are
not an intended recipient of this message, please notify the sender by
replying to this message and then delete it from your system. Use,
dissemination, distribution, or reproduction of this message by unintended
recipients is not authorized and may be unlawful.
*****************************************************************************

winsurfwon, 2018-03-16:
Is there a list of all male connectors for all the BB Blue, and is there any MFG that actually sell a set of them minus the power connectors and USB.  It is well specified in the glossy print, but not the actual male mfg connector lead part numbers if one exists.  I called a few but they were not ready for it. So good so far.

jadonk, 2020-06-09:
Issue migrated to beagleboard/beaglebone-blue#52

closed 2020-06-09 by jadonk