jadonk/beaglebone-blue #33: Adjust to fit in Altoid's tin

jadonk, 2016-10-12:


erikwelsh, 2016-11-01:
Moved power connector in as much as possible.  I think it should fit but will not know until we get the A2 boards back.

jadonk, 2016-11-02:
Still would like to find a shorter power connector.

jadonk, 2016-11-07:
I've ordered some PCBs from OSH Park to do a mechanical mock-up to verify the fix. I've not identified a source of a shorter power connector.

jonnycowboy, 2016-11-16:
Would implementing issue 39 (new BMS) allow us to use a different power connector? Is the barrel jack currently used to charge the battery or is it used for directly powering the processor?
https://github.com/jadonk/beaglebone-blue/issues/39

StrawsonDesign, 2016-11-17:
The barrel jack is primarily for charging the battery but can provide power to the 5V regulator when no battery is connected. We picked this connector because of how ubiquitous 12v wallwarts are with this 5.5mm barrel jack. I just wish there were half-length jacks available.

jadonk, 2016-12-08:
I've not been able to find a shorter jack, which is very surprising to me.

A2 still does NOT fit in an Altoids tin. :-(

jadonk, 2017-02-09:
@StrawsonDesign For rev B, can we switch the power input to a JST connector?

https://www.sparkfun.com/products/8734

I wonder how much it would cost to include that in the kit. :-/

StrawsonDesign, 2017-02-10:
Hmmmm, sorry but I really really like the universal barrel jack. We would
get so much pushback from people complaining about needing dedicated
adapters just to charge the thing just like Apple gets with their headphone
jack adapters. I do adore the Altoids tin concept, don't get me wrong! But
in practice I don't think anyone is running their board in a conductive
tin, whereas everyone will need to charge it and everyone with a wifi
router has one of these 12v dc wallwarts in their home already.

Charging via a 9v or 12v capable USB-C connector may get initial pushback
too but at least that's looking a bit more towards a future universal
standard.

Respectfully,
James

jadonk, 2020-06-09:
Issue migrated to beagleboard/beaglebone-blue#50

closed 2020-06-09 by jadonk