jadonk/beaglebone-blue #41: MPU 9250 via SPI

mirkix, 2016-12-23:
Hi, is it an option to connect the MPU 9250 via SPI and not via I2C? With Ardupilot we can sample at 4 kHz / 8 kHz (acc gyro) in this case I2C is not fast enough for the data transfer. It is possible to sample slower too, than the transfer speed via I2C should be okay.

StrawsonDesign, 2016-12-28:
Hi Mirko,

Sadly the 9250 is on the I2C bus on both Blue and Cape. In my flight code I
use the 9250's DMP for quaternion orientation estimation and also use the
DMP's fixed-rate interrupt function (up to 200hz only) to generate the
timing for the discrete feedback loop. Pulling accel, gyro, mag, and
quaternion data at 200hz puts the i2c bus at about %15 use when taking
advantage of the 9250's FIFO buffer to do one big read at a time.

I have no doubt sampling just the accel, and gyro 1khz would be possible
over I2C with intermittent reads to the magnetometer. However, even on a
small quadcopter the crossover frequency for the roll/pitch feedback
controllers is only around 10-20hz so I've found 200hz sample rate to be
more than sufficient for smooth feedback control. Anything faster is
difficult to achieve in userspace linux without multiple cores anyway.

Best,
James

Dept. Mechanical & Aerospace Engineering
University of California, San Diego

mirkix, 2016-12-28:
@StrawsonDesign Hi James, is there any documentation free available about the DMP? I think ArduPilot try to sample as fast as possible to avoid problems caused by vibrations (aliasing).
I am working on ArduPilot support for the Blue and made good progress so far.

Regards
Mirko

jadonk, 2017-01-13:
@mirkix @StrawsonDesign shall I try to ping Invensense on the DMP documentation issue?

StrawsonDesign, 2017-01-13:
Hi Mirko, Jason,

Free documentation on the DMP is limited to their example code and a
pamphlet with a brief overview of functionality. I've played with the DMP a
lot over the past few years and I think I've gotten as much performance out
of it as I can having written my implementation %75 from scratch. I even
managed to get it to package the magnetometer data into the same FIFO
packet as the DMP, accel, and gyro data which a comment in Pansenti's
mpu9250 driver said was "impossible". The sparkfun library is horrifically
slow and does a lot of unnecessary register modifications every time it
swaps between reading mag and dmp.

The DMP is internally limited to 200hz, presumably because it has to do a
LOT of floating point arithmetic for it's quaternion estimate. It always
samples and runs its filter at 200hz, even if you set up the interrupt
signal to run as a divisor of that. Sadly, that also means that the accel
and gyros have to be set to sample at 200hz too, so pinging them between
DMP reads is useless.

I understand your worry about aliasing, but 200hz is still 20x faster than
the crossover frequency of even my fastest feedback controller. A properly
designed compensator (eg not PID!) will have gain roll-off well before
200hz.

Best,
James

jadonk, 2017-01-24:
@mirkix Can you confirm you are comfortable with @StrawsonDesign's answer?

mirkix, 2017-01-25:
@jadonk :+1:

jadonk, 2020-06-09:
Issue migrated to beagleboard/beaglebone-blue#43

closed 2020-06-09 by jadonk