jadonk/beaglebone-blue #46: Getting Started

3space, 2017-04-26:
Had this board for a day.  Powered it on and opened the getting started page.  RNDIS tethering doesn't work on OS X Yosemite.  Serial connection doesn't work unless you download full drivers from FTDI.  I tried accessing beagle bone access point but don't know the password.  Was able to login through the terminal with default password, but don't know how to update software without network connection.  Now beagle bone doesn't seem to be working -  not able to browse device files in Finder window; won't connect serially; can't see access point; and RNDIS won't work.  Any other way to access beagle bone?

jadonk, 2017-04-27:
Password is "BeagleBone". Update software using http://bbb.io/latest.

3space, 2017-04-27:
How do I boot old image from flash again?

jadonk, 2017-04-27:
Remove the SD card and you'll boot from the on-board flash again.

jadonk, 2017-04-27:
For USB interface via Yosemite, have you tried 192.168.6.2?

3space, 2017-04-27:
192.168.6.2 works. What is the password to ssh into root account?

jadonk, 2017-05-16:
ssh to root is disabled. ssh as debian and then do 'sudo -s'.

OscarJGV26, 2017-09-01:
I have problems to apt-get update and git clone anything... It says connection refused on port 443. 

I am already able to ping 8.8.8.8 and www.google.com. 

I am sharing the internet through my computer with /sbin/route add default gw 192.168.7.1 (see derek maloy usb network). I do it like this and not through wifi because I can't get wifi to work on thee university network (eduroam) and I have since there is a problem with this type of networks (IEEE8021x).

Could I get some help with this please? It's driving me nuts that I can't git clone anything.

jadonk, 2020-06-09:
Issue migrated to beagleboard/beaglebone-blue#44

closed 2020-06-09 by jadonk